ccMSC/ckb #536: Mac OS Sierra

thecreativegeek, 2016-12-04:
Is there going to be any support for Mac OS Sierra anytime soon 
as i rely heavily on the option click feature in photoshop and for some reason this isn't working for me
it doesnt seem to hold the key press 

Does someone have a fix ?

jolosantos, 2016-12-08:
Concurring with @mhennessie 

I have saved static profile as my default and it works without CKB running. 

CKB have issues with mouse click + shift binding so I resorted to this for now.

BenLeggiero, 2016-12-13:
Flashing a profile to the keyboard is a workaround, not a solution. I hope @ccMSC fixes this soon.

mhennessie, 2016-12-13:
Correct but he has been completely silent for 6 months.

BenLeggiero, 2016-12-14:
Another workaround is keeping the cursor still, then holding <kbd><small>alt</small><br/>option</kbd>, then clicking. At least this works for me in Xcode, as well as with <kbd>⌘<br/>command</kbd>.

Kaurin, 2016-12-27:
If your Sierra build fails with an error message that mentions "clockid_t", take a look at [this pull request](https://github.com/ccMSC/ckb/pull/541), edit files accordingly and retry the quickinstall. That worked for me.

light2yellow, 2017-01-26:
Sierra support is very poor so far. The progress is tracked in respective issues [here](https://github.com/mattanger/ckb-next/issues).