ccMSC/ckb #537: K70 RGB Rapidfire Not Detected

mdblaz87, 2016-12-04:
I need help with this please. Have 20 days or so to return the keyboard. Running it on OSX 10.11.6

cvk14, 2016-12-05:
I can concur with this. it is not showing at all.

corporategoth, 2016-12-08:
If you use the code from the 'newdev' branch, it is detected and works perfectly (along with the M65 Pro mouse).  I am using both and they work fine.

jolosantos, 2016-12-08:
Yes. The `newdev` branch worked for me. I have the K65 Lux. I've posted several comments including the compiled `newdev` build but some still encounter issues with it. 

I hope someone fork CKB that fully works both for mac and linux.

ags911, 2016-12-08:
corporategoth where is this compiled version could you send it to me by any chance? I have compiled my own using the instructions but it will not detect my M65 Pro RGB Mouse. Is there something I have to do in the library/daemon folder or something I'm missing? I'm running OSX 10.12.1 and the M65 RGB (Normal) version was detected and worked before I returned it for the Pro.

corporategoth, 2016-12-08:
@ags911 I don't use OSX, sorry, but it is working for me on Linux.  I compiled it myself (it was dead easy, I just ran 'quickinstall').

ags911, 2016-12-15:
Oh ok thanks, no luck still.

CreedoGunn, 2016-12-25:
same here, the newdev isn't working either. and i just got this for Christmas...ouch.

ragonlan, 2016-12-30:
Same behaviour here with K70 Rapidfire and linux. Version 0.2.6. dmesg show this about corsair k70:

```
[    3.772600] hid-generic 0003:0461:4D5F.0001: input,hidraw0: USB HID v1.11 Mouse [HP Laser Gaming Mouse with VoodooDNA] on usb-0000:00:1a.0-1.4/input0
[    4.024637] hid-generic 0003:1B1C:1B38.0002: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 RGB RAPIDFIRE Keyboard ] on usb-0000:00:1a.0-1.2.1/input0
[   14.179876] hid-generic 0003:1B1C:1B38.0003: usb_submit_urb(ctrl) failed: -1
[   14.179900] hid-generic 0003:1B1C:1B38.0003: timeout initializing reports
[   14.237409] hid-generic 0003:1B1C:1B38.0003: input,hiddev0,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 RGB RAPIDFIRE Keyboard ] on usb-0000:00:1a.0-1.2.1/input1
[   14.237539] usbhid 1-1.2.1:1.2: couldn't find an input interrupt endpoint

```

giftig, 2016-12-30:
Thanks guys, I came here because I couldn't get my new K70 lux to be detected at all, heard this driver could help, had no luck, and then newdev solved my issue.

Excellent driver, turned my £150 doorstop into an excellent keyboard. 10/10

Comanda84, 2017-01-03:
same here. Please add support to the Rapidfire

ragonlan, 2017-01-03:
After download and compile `newdev` branch all start working again. The only bug i notice is daemon fails if you change `BIOS switch` to other number.

mmsz111, 2017-01-28:
K70 Rapidfire
[image]

help...plz I tried all branches