ccMSC/ckb #538: Can't Add hid device: -110 (With KVM Switch)

51systems, 2016-12-04:
This is probably a bit out of scope for the driver, but any help would be appreciated.
I've got the STRAFE (non-rgb) hooked up to a kvm shared between windows / linux. When switching to windows, everything is great, but switching back to linux, the keyboard seems to be detected, then hang for ~30 seconds. I believe at this point the kernal maybe resets the USB port, at which point the keyboard finally comes up properly.

Any suggestions are greatly appreciated.

```
[ 3218.917991] usb 1-6: new full-speed USB device number 24 using xhci_hcd
[ 3219.086776] usb 1-6: New USB device found, idVendor=058f, idProduct=9254
[ 3219.086780] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3219.086783] usb 1-6: Product: Generic USB Hub
[ 3219.086786] usb 1-6: Manufacturer: ALCOR
[ 3219.087598] hub 1-6:1.0: USB hub found
[ 3219.087669] hub 1-6:1.0: 4 ports detected
[ 3219.360987] usb 1-6.1: new full-speed USB device number 25 using xhci_hcd
[ 3219.438426] usb 1-6.1: New USB device found, idVendor=1b1c, idProduct=1b15
[ 3219.438431] usb 1-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3219.438434] usb 1-6.1: Product: Corsair STRAFE Gaming Keyboard
[ 3219.438437] usb 1-6.1: Manufacturer: Corsair
[ 3219.438439] usb 1-6.1: SerialNumber: 0302901AA
[ 3219.440354] input: Corsair Corsair STRAFE Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/0003:1B1C:1B15.0019/input/input50
[ 3219.440582] hid-generic 0003:1B1C:1B15.0019: input,hidraw0: USB HID v1.11 Keyboard [Corsair Corsair STRAFE Gaming Keyboard] on usb-0000:00:14.0-6.1/input0
[ 3255.800765] usbhid 1-6.1:1.1: can't add hid device: -110
[ 3255.800803] usbhid: probe of 1-6.1:1.1 failed with error -110
[ 3255.800989] usbhid 1-6.1:1.2: couldn't find an input interrupt endpoint
[ 3255.864638] usb 1-6.2: new full-speed USB device number 26 using xhci_hcd
[ 3255.946435] usb 1-6.2: New USB device found, idVendor=062a, idProduct=4101
[ 3255.946437] usb 1-6.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3255.946439] usb 1-6.2: Product: 2.4G Keyboard Mouse
[ 3255.946439] usb 1-6.2: Manufacturer: MOSART Semi.
[ 3255.947701] input: MOSART Semi. 2.4G Keyboard Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.2/1-6.2:1.0/0003:062A:4101.001A/input/input51
[ 3256.000147] hid-generic 0003:062A:4101.001A: input,hidraw1: USB HID v1.10 Keyboard [MOSART Semi. 2.4G Keyboard Mouse] on usb-0000:00:14.0-6.2/input0
[ 3256.001972] input: MOSART Semi. 2.4G Keyboard Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.2/1-6.2:1.1/0003:062A:4101.001B/input/input52
[ 3256.031794] hid-generic 0003:1B1C:1B15.0019: usb_submit_urb(ctrl) failed: -1
[ 3256.032427] input: ckb1: Corsair STRAFE Gaming Keyboard as /devices/virtual/input/input53
[ 3256.083952] hid-generic 0003:062A:4101.001B: input,hiddev0,hidraw0: USB HID v1.10 Mouse [MOSART Semi. 2.4G Keyboard Mouse] on usb-0000:00:14.0-6.2/input1
[ 3256.084661] input: ckb1: Corsair STRAFE Gaming Keyboard as /devices/virtual/input/input54
[ 3256.255683] usb 1-6.3: new low-speed USB device number 27 using xhci_hcd
[ 3256.338866] usb 1-6.3: New USB device found, idVendor=10d5, idProduct=55a2
[ 3256.338867] usb 1-6.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3256.338868] usb 1-6.3: Product: 2Port KVMSwicther
[ 3256.338869] usb 1-6.3: Manufacturer: No brand
[ 3256.338870] usb 1-6.3: SerialNumber: 02\xffffffc2\xffffff92
[ 3256.346411] input: No brand 2Port KVMSwicther as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0/0003:10D5:55A2.001C/input/input55
[ 3256.398453] hid-generic 0003:10D5:55A2.001C: input,hidraw2: USB HID v1.10 Keyboard [No brand 2Port KVMSwicther] on usb-0000:00:14.0-6.3/input0
[ 3256.399499] usbhid 1-6.3:1.1: couldn't find an input interrupt endpoint
[ 3256.596816] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110
[ 3261.972807] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
[ 3267.112848] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
[ 3272.252779] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
[ 3277.392695] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
[ 3282.532581] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
[ 3283.176481] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110
[ 3283.688480] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110
[ 3284.204468] usb 1-6.1: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110

tatokis, 2016-12-04:
https://github.com/ccMSC/ckb#troubleshooting

51systems, 2016-12-04:
Tried putting `usbhid.quirks=0x1B1C:0x1B15:0x20000408` in grub with no change in behaviour. Unfortunately I'm using FDE, so I can't try the `0x4` option. Additionally, usbhid is builtin, so I can't reload it with the `0x4` option after boot.

If I pull the USB after switching the kvm switch, then plug the KB back in it seems to immediately pick it up. It seems to me that there might be a race condition when the kvm switches over. If I could delay the probing of the keyboard, say for 1 second, that might make it work.

51systems, 2016-12-04:
Seem to have gotten it working by switching the ports on the KVM the computers were plugged into, plus switching the USB port that the switch was plugged into on the linux box. Perhaps it was a power issue.

Thanks for the help!

closed 2016-12-04 by 51systems

tatokis, 2016-12-04:
Glad to hear you got it working :)