ccMSC/ckb #539: Project ERROR: Xcode not set up properly. You may need to confirm the license agreement by running /usr/bin/xcodebuild.

xeenuhs, 2016-12-10:
Keep getting this error:

Project ERROR: Xcode not set up properly. You may need to confirm the license agreement by running /usr/bin/xcodebuild.

iMac Late 2013 Version 10.12.1 (os Sierra)

Ive researched all I've could and no solution.

QT 5.7
xCode 8.1

xeenuhs, 2016-12-10:
[image]

Update. Downloaded QT 5.6 and reinstalled Xcode and that go it to install now i am receiving the errors in the picture...

mackermans, 2016-12-12:
@xeenuhs Worked for me after I killed the ckb-daemon process and started ckb.app.

batchpert, 2016-12-12:
Im currently on ckb v0.2.6+t04 and am trying to update to the newer version by tatokis. I get the same 'Project ERROR' message and am not sure what to try. I have reinstalled QT 5.6 and attempted the manual download but i got the same message. Will try re-downloading Xcode see what happens.
Any other ideas would be greatly appreciated

tatokis, 2016-12-12:
Did you set the path to your Xcode.app? 

There's a command to do so, can't remember it off the top of my head.

cvk14, 2016-12-12:
xcode-select is the command.. that doesn't change anything on my end, which if you run the -p argument, it shows the path and its correct. I tried the --install to install the command lines tools and still no luck. same error.

batchpert, 2016-12-13:
Its all good i re-downloaded Xcode and then realised i had QT 5.7 so wasn't sure if that made a difference. Downloaded 5.6 and then it all worked smoothly

mchampanis, 2017-03-09:
@xeenuhs @cvk14 I had a similar problem on a different project (the license error), and fixed it like so:

```
cd /Applications/Xcode.app/Contents/Developer/usr/bin/
sudo ln -s xcodebuild xcrun
```

See: http://stackoverflow.com/questions/33728905/qt-creator-project-error-xcode-not-set-up-properly-you-may-need-to-confirm-t