ccMSC/ckb #543: Systemd stop service & Macro Keys/Animations not working on boot

zimmertr, 2016-12-15:
Hello,

I use Arch Linux, have installed CKB, and have enabled the ckb-daemon service unit. 

When my computer boots up my keyboard works, illuminates from memory, etc. However, macro keys and animations do not work until I manually launch the ckb program. Despite the fact that ckb-daemon is already running.

Additionally, when I power off my computer, ckb-daemon fails to stop correctly and Systemd has to terminate the process after it times out.

GarlandKey, 2016-12-21:
Dev hasn't been working on this project in some time (check the dates).  We need someone who can code to fork this and continue on with the version roadmap.  Would be nice to see CUE files bumped to next priority.  I'd be willing to fund this this bounty to anyone reading and I'd be happy to promote/help with any fund raising effort.

I'd work on this myself but I'm not experienced enough to tackle it in a timely manner.

Anyway, this isn't really a bug but just a circumstance of how the software operates AFAIK.  I have a K70 RGB with the same behavior (I also use Arch).

zimmertr, 2016-12-21:
Interesting, I just figured it should be pointed out. It pisses my girlfriend off who has a hard enough time using my computer already since I use a window manager and use hotkeys/terminal commands to launch programs.

I would volunteer but I'm more of a DevOps engineer than an Engineer. Although this would be a fun project with which I could strengthen my skills.

mattanger, 2016-12-22:
We've got another fork set up and its under active development. Right now
we're working on bug fixing and merging lingering PRs. No support right
now, but testing and technical info is welcome. Once we get stable and
figure out a new name we'll do a release. The fork can be found in my
profile for anyone interested.

zimmertr, 2016-12-22:
Awesome, great news. Glad to hear that @mattanger

tatokis, 2016-12-22:
The UI is required to be running in order for ckb-daemon to do anything 
other than pass through keystrokes. The quickinstall script uses xdg to 
install a .desktop file that gets launched on system boot.

Regarding systemd, I believe that the same issue has been reported 
before. I run
```
$ systemd --version
systemd 229
+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP 
+LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ -LZ4 +SECCOMP +BLKID +ELFUTILS 
+KMOD -IDN
```
and I have not been able to reproduce this.

zimmertr, 2016-12-22:
If I close the UI the macro keys and animations continue to work. Even in the process is killed.

tatokis, 2016-12-22:
The animations are separate executables launched by ckb.
[image]

zimmertr, 2016-12-22:
Ahh, cool. Thanks for the info!