ccMSC/ckb #544: quickinstall warning report

wanghran, 2016-12-16:
I downloaded the whole file and tried to use quickinstall. Then, two warning generated. Could you please help me on that. My mac system is 10.12.1 and my device is K70 RGB Rapidfire.

./keyaction.h:96:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:97:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
In file included from release/moc_mperfwidget.cpp:9:
In file included from release/../mperfwidget.h:11:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:96:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:97:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
In file included from release/moc_kperfwidget.cpp:9:
In file included from release/../kperfwidget.h:9:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:96:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:97:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
2 warnings generated.
2 warnings generated.
2 warnings generated.

GarlandKey, 2016-12-21:
Dev is AWOL.  I'd help but I don't use Apple products.

wanghran, 2016-12-21:
Thank you for replying. Do you have any suggestion that I can start with to solve this problem?

tbcrawford, 2016-12-22:
I do believe this spawns from `os.h` in the ckb-daemon not being able to first find the Carbon/Carbon.h file. I've been trying to investigate this myself on macOS Sierra but can't determine how to get Qt to look in a different or specific spot for all of the frameworks available. You should be able to find your available frameworks at `/System/Library/Frameworks/` and in here Carbon.framework is clearly available so I'm not quite sure what the issue is. If anyone has any further tips knowing this, I'd love to test them out.

mattanger, 2016-12-22:
We've got another fork going and we're looking to fix this issue. We're a
bit lacking with OSX, and I personally don't have access to Sierra. But if
we've got people to test we can work on implementing some fixes to get
support for Sierra going. You can find the fork here
https://github.com/mattanger/ckb. Feel free to post anything that you've
found in regards to this issue. Right now, we're not offering any sort of
support, just trying to get all the pull requests implemented and any fixes
done before we do any sort of release.

tbcrawford, 2016-12-22:
@mattanger I saw that there was a new fork going under your username actually! I would be happy to test where I can. I will be sure to start your fork to keep up with it. On another note, I do believe anything after the `#include <Carbon/Carbon.h>` may have a problem as well when trying to find the library. The next lines reference `IOKit`. I may try to reinstall El Capitan on another drive I have to verify that none of the frameworks moved on macOS. Unless someone else could verify this for me that sees this. Currently on macOS Sierra, they are located at `/System/Library/Frameworks/`. I don't fully believe they would have moved from one OS to another but I'm just trying to start finding where the error may be. If I start a new project however in Qt and try to include Carbon, it compiles just fine, so I'm thinking there is something wrong with the project itself. Possibly in the pro files (which I believe are qmake files). Sorry, I'm new to Qt and trying to learn as I go.

Also, if looking under `/Applications/Xcode.app/Contents/Developer/` for the `Carbon.h` header files, they can be found here: 

```
/Applications/Xcode.app/Contents/Developer//Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/Carbon.framework/Versions/A/Headers/Carbon.h
/Applications/Xcode.app/Contents/Developer//Toolchains/XcodeDefault.xctoolchain/usr/lib/swift-migrator/sdks/MacOSX.sdk/System/Library/Frameworks/Carbon.framework/Versions/A/Headers/Carbon.h
```

To replicate finding this, use the following command: `find /Applications/Xcode.app/Contents/Developer/ | grep Carbon.h`