ccMSC/ckb #546: device not connected

iamjustgodly, 2016-12-24:
ITS A k70 rgb lux and its says device not connected

CreedoGunn, 2016-12-25:
same here with my K70 RGB Rapidfire

CreedoGunn, 2016-12-25:
NEVERMIND, i fixed it (lol i'm a complete noob) run the ckb-dameon as root in your terminal then launch it. works like a charm.

ximiii, 2016-12-26:
CreedoGunn , Can you post a video or write a step by step process from download to finish in 
"Noob Terms" showing how to do this? I apologize for being a noob at doing this but i would really like to get this thing to work. I have tried the following . 

Machine:
Mac mini (late 2012 ) 2.6 i7 
Mac os Sierra V: 10.12.1

Keyboard : 
Corsair k70 RGB Rapid Fire 
( Note, I updated the keyboard on windows and it works fine on osx just cant mess with the CKB Config options for color and effects )

First Attempt :
Downloaded the ckb.pkg from this link https://github.com/ccMSC/ckb/releases/tag/v0.2.6
 
went through the dialogue , CKB installed into my applications . Launched it Got a " no device connected " 

Second Attempt:

Downloaded xcode 
Downloaded qt5  
Downloaded Source code from this link https://github.com/ccMSC/ckb/releases/tag/v0.2.6
 
Unzipped ckb 0.2.6.zip file and clicked quickinstall and got the following messsage on terminal 
(was gonna post full terminal message but it was very long did not want to make this message longer than needed. heres a brief view of it.

./os.h./os.h::47:10: 47:10: fatal error: fatal error: 'Carbon/Carbon.h' file not found
'Carbon/Carbon.h' file not found

2 warnings generated.
2 warnings generated.
2 warnings generated.
Press enter to exit.

Still Said No Device Detected 

Attempt #3 

Downloaded New Dev zip file  

I did the same steps as Attempt #2 got the same results from terminal, but this time ckb says driver version mismatch (beta-v0.2.6) Please upgrade ckb-daemon. if problem presists try rebooting. 

Clearly i am doing something wrong as its working for others except me and a few other people who have no in depth experience with Terminal or how to compile applications to work. 

Thanks Again.

legoman1119, 2016-12-26:
Same things happened to me ximii. I'm utterly confused on how to get this to work.

ximiii, 2016-12-27:
Hopefully somebody will chime in soon

ximiii, 2016-12-28:
@iamjustgodly @legoman1119 Look under my post i created . @tatokis  helped get mine to work see if what i did helps you guys . Cheers :)

legoman1119, 2016-12-28:
Thanks I'll take a look at it