ccMSC/ckb #549: Can Anybody help me with CKB ?

ximiii, 2016-12-26:
Can anybody  post a video or write a step by step process from download to finish in
"Noob Terms" showing how to do this? I apologize for being new at doing things like this but i would really like to get this thing to work. I have tried the following .

Machine:
Mac mini (late 2012 ) 2.6 i7
Mac os Sierra V: 10.12.1

Keyboard :
Corsair k70 RGB Rapid Fire
( Note, I updated the keyboard on windows and it works fine on osx just cant mess with the CKB Config options for color and effects )

First Attempt :
Downloaded the ckb.pkg from this link https://github.com/ccMSC/ckb/releases/tag/v0.2.6

went through the dialogue , CKB installed into my applications . Launched it Got a " no device connected "

Second Attempt:

Downloaded xcode
Downloaded qt5
Downloaded Source code from this link https://github.com/ccMSC/ckb/releases/tag/v0.2.6

Unzipped ckb 0.2.6.zip file and clicked quickinstall and got the following messsage on terminal
(was gonna post full terminal message but it was very long did not want to make this message longer than needed. heres a brief view of it.

./os.h./os.h::47:10: 47:10: fatal error: fatal error: 'Carbon/Carbon.h' file not found
'Carbon/Carbon.h' file not found

2 warnings generated.
2 warnings generated.
2 warnings generated.
Press enter to exit.

Still Said No Device Detected

Attempt #3

Downloaded New Dev zip file

I did the same steps as Attempt #2 got the same results from terminal, but this time ckb says driver version mismatch (beta-v0.2.6) Please upgrade ckb-daemon. if problem presists try rebooting.

Clearly i am doing something wrong as its working for others except me and a few other people who have no in depth experience with Terminal or how to compile applications to work.

Thanks Again.

RushanB, 2016-12-26:
I have the same issue as you mentioned with the same errors showing.

ximiii, 2016-12-27:
Hopefully somebody will chime in soon

RushanB, 2016-12-27:
I decided to reutrn the mouse and get a Logitech one instead as the drivers support Mac OS.

chrisg123, 2016-12-27:
ximiii can you confirm that the `VERSION` file inside the newdev zip you downloaded contains beta-v0.2.6+d01 ?

tatokis, 2016-12-27:
Try compiling this https://github.com/mattanger/ckb

tatokis, 2016-12-27:
Forgot to add, requires Qt 5.6, NOT 5.7.

ximiii, 2016-12-28:
@tatokis
My Qt says : 
Qt Creator 4.2.0 (4.2.0)
Based on Qt 5.7.1 (Clang 7.0 (Apple), 64 bit)
Built on Dec 9 2016 12:24:11
From revision 5071dd80cb

I am now downloading qt Qt 5.6.2 for macOS (1.1 GB)  

Link in "read me" sends you to the main open source page : 
https://www.qt.io/download-open-source/

did not read i was downloading 5.7 i am now downloading 5.6 will chime back in a little bit and will retry my attempts 1-3 and then the link you posted and i will tell you know if it works 

@chrisg123  ckb beta-v0.2.6+d01 in the newdev zip file

ximiii, 2016-12-28:
@tatokis 
My Qt now says :

 Qt Creator 4.0.3
Based on Qt 5.6.1 (Clang 7.0 (Apple), 64 bit)
Built on Jul 5 2016 04:18:36
From revision ce4ddcb060

I did my " Attempts 1 - 3 " same errors only thing that changed is the ckb from new dev now says no device detected instead of " driver version mismatch (beta-v0.2.6) Please upgrade ckb-daemon. if problem presists try rebooting "

i downloaded your zip which was ckb-master.zip . Unzipped clicked quickinstall got errors on terminal. clicked ckb.app in the folder   " ckb beta-v0.2.7 "  did restarts each time . plugged out usb plugged back in still says  device not detected........

tatokis, 2016-12-28:
What kind of errors did you get from the one I linked you?

ximiii, 2016-12-28:
@tatokis heres the full terminal when hitting quickinstall

Unknown:~ Ximiii$ /Users/Ximiii/Desktop/ckb-master/quickinstall ; exit;
readlink: illegal option -- f
usage: readlink [-n] [file ...]
Preparing build files...
Compiling binaries...
(This can take a while, please be patient)
cd src/ckb-ripple/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb-ripple/ckb-ripple.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ckb-wave/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb-wave/ckb-wave.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ckb-gradient/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb-gradient/ckb-gradient.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ckb-random/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb-random/ckb-random.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ckb-rain/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb-rain/ckb-rain.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ckb-pinwheel/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb-pinwheel/ckb-pinwheel.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ckb/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb/ckb.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ckb-daemon/ && ( test -e Makefile || /Users/Ximiii/Qt/5.7/clang_64/bin/qmake /Users/Ximiii/Desktop/ckb-master/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -c -pipe -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.8 -std=c99 -O2 -fPIC  -I. -I../../../../Qt/5.7/clang_64/mkspecs/macx-clang -o main.o main.c
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -c -pipe -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.8 -std=c99 -O2 -fPIC  -I. -I../../../../Qt/5.7/clang_64/mkspecs/macx-clang -o main.o main.c
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -c -pipe -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.8 -std=c99 -O2 -fPIC  -I. -I../../../../Qt/5.7/clang_64/mkspecs/macx-clang -o main.o main.c
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -c -pipe -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.8 -std=c99 -O2 -fPIC  -I. -I../../../../Qt/5.7/clang_64/mkspecs/macx-clang -o main.o main.c
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -c -pipe -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.8 -std=c99 -O2 -fPIC  -I. -I../../../../Qt/5.7/clang_64/mkspecs/macx-clang -o main.o main.c
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile.Debug
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -c -pipe -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.8 -std=c99 -O2 -fPIC  -I. -I../../../../Qt/5.7/clang_64/mkspecs/macx-clang -o main.o main.c
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /Users/Ximiii/Desktop/ckb-master/src/ckb-daemon/../../VERSION`\" -I. -I../../../../Qt/5.7/clang_64/mkspecs/macx-clang -o debug/device.o device.c
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile.Release
/Users/Ximiii/Qt/5.7/clang_64/bin/uic mainwindow.ui -o ui_mainwindow.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic kbwidget.ui -o ui_kbwidget.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic settingswidget.ui -o ui_settingswidget.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic kblightwidget.ui -o ui_kblightwidget.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic kbanimwidget.ui -o ui_kbanimwidget.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic animadddialog.ui -o ui_animadddialog.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic animsettingdialog.ui -o ui_animsettingdialog.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic gradientdialog.ui -o ui_gradientdialog.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic kbprofiledialog.ui -o ui_kbprofiledialog.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic kbbindwidget.ui -o ui_kbbindwidget.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic rebindwidget.ui -o ui_rebindwidget.h
/Users/Ximiii/Qt/5.7/clang_64/bin/uic modeselectdialog.ui -o ui_modeselectdialog.h
[... cut ...]
./keyaction.h:164:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:165:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
2 warnings generated.

Press enter to exit.
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

tatokis, 2016-12-28:
Install the xcode command line tools and try again.

ximiii, 2016-12-28:
xcode-select --install     

do i use this in the terminal ?

tatokis, 2016-12-28:
Yes

ximiii, 2016-12-28:
@tatokis downloading now . ill report in a few mins.

ximiii, 2016-12-28:
@tatokis , it worked beautifully with no issues except for i did not have a password set on my mac and the terminal did not like that , so i set a password and it finished installing by pressing the y , y ,y in terminal as ckb is finishing its download. also i gave it a restart .  no issues what so ever so far, thank you for taking the time to help me and im sure others will read this and be educated by you. 

[image]

SNARKYPOTATO, 2017-01-17:
@tatokis Please help, I've spent all day try8ing to get the blasted thing to work. I'm getting tonnes of warning messages in terminal.