ccMSC/ckb #550: M65 Pro RGB does not show up

RushanB, 2016-12-26:
Running on Mac OS X El Capt. 10.11.6.

"No Devices Connected" when downloaded from package.

When trying to compile with QT 5.7 error comes up with Carbon.h

When trying to compile with QT.5.6 error comes up with define Clock_t
-After commenting out the error line, code compiles and runs
-Again shows "No Devices Connected"

Need someone to fix this please.

RushanB, 2016-12-27:
Returned and got a Logitech device instead but still would like to see someone fix the code.

sh4d0w28, 2016-12-30:
@RushanB , Please, try `newdev` branch. They have support for your keyboard in there. Looks like developer paused project :(

mattanger, 2016-12-30:
We've got a new fork of the ckb project going, and we're working on
incorporating bugfixes and improvements. It can be found here:
https://github.com/mattanger/ckb-next
We haven't made a stable release, and we're not offering support until we
do a stable release. But, some improvements can be found in our fork, and
we're open to testing/usage feed back. Just no help me issue, keep those to
this project please! When possible one of us may chime in here and offer
what information we can.

sh4d0w28, 2016-12-31:
@mattanger , do you have any connection with @ccMSC ? Or you just created fork? I`m asking as we have 160 forks now

tatokis, 2016-12-31:
There's no connection. 
We just decided to team up in order to pick up development from where it was left off and continue with the project. 

Most likely, the rest of the forks were created to implement a feature and submit a pull request.

sh4d0w28, 2016-12-31:
@tatokis so i can transfer my pull request to new repo, right?

tatokis, 2016-12-31:
You don't need to, I can merge it without you making a new one.
I just wanted to test it first, make sure that it's stable and it works 
before merging it.