ccMSC/ckb #552: Keyboard lights up but only shows settings tab in CKB

nmann18, 2016-12-29:
After installing CKB i restarted my computer about three times and it only shows the settings tab. (Keyboard is plugged in and keys are lit up). Any solutions to this?

nmann18, 2016-12-29:
Also i followed troubleshooting steps still not working. If anyone could get back to me that would be great.  Hope i can get this working.

sh4d0w28, 2016-12-30:
What keyboard do you use? 
I had K70 RGB LUX with same mistake.
To solve i just switched to `newdev` branch. 
Hope it will help you as well.

nmann18, 2016-12-30:
i am also using the the k70. But i'm on OSX. What is a "newdev" Branch? @sh4d0w28 
ps, thanks for getting back to me!

sh4d0w28, 2016-12-30:
@nmann18 , https://github.com/ccMSC/ckb/tree/newdev - this one. So you need to build project from it.

mattanger, 2016-12-30:
> @nmann18 <https://github.com/nmann18> , https://github.com/ccMSC/ckb/
> tree/newdev - this one. So you need to build project from it.

For anyone who's interested, we're working on a fork that's incorporating a
lot of the improvements that haven't been added to this project. It can be
found here: https://github.com/mattanger/ckb-next. It's not "stable" yet,
so we're not offering support yet, but a few lingering issues have been
resolved. We're also working on improved OSX support.

nmann18, 2016-12-30:
@sh4d0w28 ok so i tried using the binary download on the newdev branch and still hasn't worked