ccMSC/ckb #557: Dead Project? - CKB won't open

t3kg33k, 2017-01-06:
So, is this a dead project? It would be unfortunate if so.
I'm using Ubuntu Mate 16.04. At some point over the past few months the application stop displaying in my tray and it will not open when I select it from the Application menu. Some update must have broke it along the way.
Any way to fix this? I sure would love to change my color scheme but can't since the application won't open.

mattanger, 2017-01-06:
There is an active fork, https://github.com/mattanger/ckb-next but no
support right now.

Easy question, but does ckb work when running manually?

t3kg33k, 2017-01-06:
What do you mean when you say "running manually"?

mattanger, 2017-01-06:
I mean executing the ckb binary, and/or manually restarting the daemon.

t3kg33k, 2017-01-07:
I restarted the systemd service with sudo systemctl restart ckb-daemon.service command and attempted to start the application from the Application menu and it still does not open.

mattanger, 2017-01-07:
Have you tried running the client from the terminal?

t3kg33k, 2017-01-07:
Yes. It tells me that ckb is already running:

[ ed@edmate1604: ~ ]$  ckb
ckb is already running. Exiting.

frickler24, 2017-01-08:
ok, then use in your terminal-program the following line

`ckb --close ; sleep 5; ckb &` 

This will close your current running implementation first and then start a new one.
(tested on linux, you may reduce the sleeping time if you have a fast machine :-))

t3kg33k, 2017-01-09:
That worked. Thanks.