ccMSC/ckb #560: STRAFE non-rgb - Backslash key (above enter key) will not light up at all

Inkpen1000, 2017-01-13:
This software is great and all but now my backslash key won't light up at all.

I assume it's nothing serious and can be fixed with a slight modification to the source code but right now it's just really annoying.

p.s. I'm pretty sure that the LED under the key isn't damaged because I got this keyboard just a few days ago and it was working fine before I used the software. However, if there are other STRAFE users out there who don't experience this problem then maybe it's something on my end after all :P

Thanks,
Logan

jennydaman, 2017-01-23:
Sounds like a strange problem, I have a non-RGB STRAFE and it works fine with ckb. What OS are you on? I had ckb installed manually on Ubuntu 14.04, though right now, I'm on Antergos, and it works fine. I suggest you try reinstalling ckb, checking the keyboard settings (if you click on a key, you will change an individual key's settings, not the keyboard's settings!), or update the keyboard's firmware.

mattanger, 2017-01-23:
That is indeed a strange problem. Have you verified that it works in
Windows or have you verified the led works by disabling ckb-daemon? It
should just load with default lighting scheme. Also, there's a new fork:
https://github.com/mattanger/ckb-next that's under active development.
@ccMSC seems to have abandoned this project.

Inkpen1000, 2017-01-24:
After a while, I just decided to return the keyboard as there were multiple other problems with the it to begin with. Thanks for your suggestions though!

closed 2017-01-24 by Inkpen1000