ccMSC/ckb #562: Cut a release - CKB has been rather stable now

linjer, 2017-01-15:
Hey there. Been using CKB for a while and really appreciate this work. I've used it through many versions now, and as of the latest version, it's been stable and works without many issues.

Perhaps it would make sense to set a `1.0` milestone for release that is no longer marked as pre-release beta? I'm open to helping.

frickler24, 2017-01-18:
@linjer, because @ccMSC isn't online since more than 12 year, plz follow the fork 
mattanger/ckb-next [https://github.com/mattanger/ckb-next](url)

linjer, 2017-01-18:
@frickler24 Can you clarify what you mean by "@ccMSC isn't only since more than 12 year"? It shows he's actively committed to ckb in June 2016. That's about 8 months ago.

frickler24, 2017-01-18:
@linjer sorry, typo: I ment 1/2 year.

clodal, 2017-02-06:
Link to the new fork is broken. This is the correct one: [ckb-next](https://github.com/mattanger/ckb-next)

closed 2018-05-11 by linjer