ccMSC/ckb #563: CKB-DAEMON consumes 25% power of the CPU after start from hibernate/sleep mode

mrstanlez, 2017-01-16:
This is a problem related to my keyboard K70 RGB. OS is Linux 64 bit Manjaro XFCE 4.8(icon in systray dissapears always after start up from hibernate/sleeping mode).

CPU is used 25% of power with difference with normal start only 1% maximum. The process is ckb-daemon the consumes so more cpu power.

Please can You fix this issue ?

algtmatt, 2017-01-16:
I've had the same high cpu usage issue with that keyboard, https://github.com/mattanger/ckb-next fork seems to be a bit better.

light2yellow, 2017-01-19:
I'm afraid there's no easy solution for Linux. MacOS has [this](https://github.com/ccMSC/ckb/blob/master/src/ckb/main.cpp#L15) line which can be commented out.

mrstanlez, 2017-01-19:
I dont think so. Because after normal start all is under 1% power usage.
So only we must think, how to restart ckb-daemon like it was from normal
start.
Correct ?

light2yellow, 2017-01-19:
I'm not yet aware of ckb codebase, this string was just what I found so far. And I didn't find any similar one for Linux. By "no easy solution" I mean without extensive refactoring of the codebase.

tatokis, 2017-01-19:
To me it sounds like after waking up, something goes wrong with the daemon, a pthread goes loose and starts taking up 100% of one core.
@mrstanlez is your CPU a quad core?

mattanger, 2017-01-20:
That is strange. So, for sure, does this happen when resuming from sleep
and hibernation? Or more often with one or the other?

mrstanlez, 2017-01-20:
Hi to All,
yes if the system resumes from sleep or hibernation. I resume with vlc and
thunar, but only ckb-daemon will comsume a lot of power cpu. Ckb
application is good, but this problem still exists. Only one solution is
kill ckb-daemon and restart ckb with keyboard re-plug from USB port 3.0.

frickler24, 2017-01-21:
@mrstanlez: To get an idea what the daemon is running, please give us some more infos:
1. running top or similar cmd, find out the PID of the cpu consuming process.
2. then run "sudo strace -p <PID-YOU-HAVE-FOUND> > /tmp/foo
3. after 1-2 seconds abort it by pressing CTRL-C
4. Have a look at the file /tmp/foo
I think  you will read endless loops of the same steps (open a file or select or something similar)
5. Copy some of that loops into this issue here or attach the whole file (may be huge, depending on your CPU speed :-)).