ccMSC/ckb #564: Corsair K70 Not found by CKB

SNARKYPOTATO, 2017-01-16:
The application will not respond to the keyboard at all.
Please help!

charendoff, 2017-01-17:
im having the same issue

SNARKYPOTATO, 2017-01-18:
Hi @charendoff, if your issue is like mine, try this fix:
Go to [here](https://github.com/ccMSC/ckb) and go to the green button that says clone & download, do that then run the ./quickinstall!

darthdeus, 2017-02-05:
I'm having the same issue. Tried building [ckb-git-latest](https://aur.archlinux.org/packages/ckb-git-latest/) but that fails to detect it too.

tatokis, 2017-02-05:
@darthdeus as also mentioned in the latest comment in the AUR page.
We're working on a fork that incorporates many fixes (and adds support for new K70 keyboards) https://github.com/mattanger/ckb-next

Give it a try

light2yellow, 2017-02-05:
https://github.com/mattanger/ckb-next/issues/5

darthdeus, 2017-02-07:
@tatokis works perfectly, thanks!

light2yellow, 2017-02-18:
@all there are ckb-next packages available in the AUR now.
https://aur.archlinux.org/packages/ckb-next-git
https://aur.archlinux.org/packages/ckb-next-latest-git