ccMSC/ckb #566: Problem with mac, el capt 10.11.6

scootcam, 2017-01-18:
When i loaded ckb.pro into qt5 and then when i went to build it, it said that a "carbon" thing wasnt working

JarrenJ, 2017-01-19:
same, lol i just opened an issue about it last night :P

scootcam, 2017-01-19:
have you found any solutions for it

JarrenJ, 2017-01-19:
Unfortunately no. I have tried re-downloading files, uninstalling and restarting, etc. My guess is that theres a problem with the newdev branch because thats where I download it from, yet nothing changes. -- So, no. no fix yet.

JarrenJ, 2017-01-21:
scootcam, I found a solution. Download the cmake folder from here: [https://github.com/mattanger/ckb-next/tree/cmake](https://github.com/mattanger/ckb-next/tree/cmake)  and run quickinstall. It might ask you to download more tools for xcode, if so say yes as you will need them. If you need any additional help just reply back to me.

light2yellow, 2017-01-21:
@scootcam @JarrenJ the `cmake` branch is the development one and is not ready to be used, please use the `testing` one next time. This is just the coincidence that both `cmake` and `testing` branches have the same files that won't break your setup, but still. The project is going to refactor the Readme file for easier troubleshooting and current project status.

JarrenJ, 2017-01-21:
Thanks for the info.