ccMSC/ckb #567: K70 stops working every time computer wakes from sleep

krishunt, 2017-01-21:
I'm using K70 RGB keyboards and ckb with Yosemite (10.10.5) on one iMac and El Capitan on another. Everything works great on El Capitan, but on the Yosemite Mac, I have to flip the polling switch every time the computer wakes from sleep; otherwise, the keys randomly type double characters, and modifier keys don't work. Is there a fix for this? Is it strictly a Yosemite issue?

krishunt, 2017-01-21:
Never mind — I moved the keyboard plug from a USB hub to the back of the iMac, and now it works fine. I guess that's the way I have it on the other iMac.

krishunt, 2017-01-31:
I tried switching the keyboards, and the problem persists on the Yosemite iMac. So it's not a keyboard issue; it sounds like an OS issue.

byinarie, 2017-03-20:
Just letting you know there is a new fork for this project :) https://github.com/mattanger/ckb-next

also see: https://github.com/mattanger/ckb-next#what-happened-to-the-original-ckb