ccMSC/ckb #570: CKB Mac 0.2.6 disappearing profiles

snowcamo, 2017-01-25:
More often than not, my profiles will disappear on a restart. I was just wondering if someone could point me to the location they are saved on Mac, so I can try and troubleshoot. Thanks!

light2yellow, 2017-01-26:
This issue was fixed in the [fork](https://github.com/mattanger/ckb-next). At least it works for me now.

snowcamo, 2017-01-26:
Using the 0.2.4 binary on that fork? Or are you building from the source? Thanks. (Sorry a little unfamiliar with github, if I misunderstood something. I wasn't even sure what a fork was at first :) )

light2yellow, 2017-01-26:
No problem. Currently we aren't shipping any binaries (but will be ASAP). For now open the Terminal and execute:
```
git clone https://github.com/mattanger/ckb-next.git && cd ckb-next && ./quickinstall
```

snowcamo, 2017-01-27:
Seems to work so far! Thanks for the help :) 👍