ccMSC/ckb #581: K95: Not recognizing hardware when replugged

ekimlinger, 2017-02-22:
Hi All,

Not sure if this a bug that has been submitted yet (I couldn't find one) but I just wanted to see if anyone has figured out a fix.

Every time (consistently) that I unplug my K95 RGB, the software doesn't recognize that it's plugged in and only shows the settings menu and it does not show that it is connected. Kind of annoying, but still part of the problem. 

The real issue is that it does not allow me to type anything. At all. And all of color profiles do not appear properly until I reboot. 

I am running el Capitan (OS X 10.11.4), MacBookPro 12,1 (early 2016 model). Just recently installed ckb. 

Is there something that I can run in order to correct this?

frickler24, 2017-02-22:
@ekimlinger under [ckb-next](https://github.com/mattanger/ckb-next) you may find more recent builds and some discussions about similar failure-situations

ekimlinger, 2017-02-24:
Yeah, that would work and I would be totally fine downloading all of the source and compiling it myself but I think they make you install Qt which takes up like 20gigs of space, which I am not okay with. I think I'm going to read up on it a little more. Have you tried it at all @frickler24 ?

frickler24, 2017-02-24:
@ekimlinger I do not understand your point of view, because ckb-next is just a fork of ckb. Both clients are based on Qt and must be compiled for the destination system. So where is the difference?

And yes, I did some development for ckb and because this development stopped suddenly, I'm happy to cooperate now with the ckb-next-team.

ekimlinger, 2017-02-25:
Well yeah they're both based off qt but the ckb-next does not have a precompiled .pkg file like the old repo had: https://github.com/ccMSC/ckb/releases/tag/v0.2.6

Is that somewhere on new repo and I'm missing something or do they not have that yet? I'd rather not download a 20 gig compiler when the app is going to only be a few megs hopefully.

byinarie, 2017-03-20:
Just letting you know there is a new fork for this project :) https://github.com/mattanger/ckb-next

also see: https://github.com/mattanger/ckb-next#what-happened-to-the-original-ckb