ccMSC/ckb #589: pt-BR layout not supported

rainerborene, 2017-04-03:
This is the one I'm currently using.

[K70 Lux Red Led Cherry MX Red - CH-9101020-BR](http://www.kabum.com.br/produto/82068/teclado-corsair-gaming-mecanico-k70-lux-red-led-cherry-mx-red-ch-9101020-br/?tag=teclado%20corsair)

tatokis, 2017-04-03:
That looks like the standard en-US layout. Also, might wanna check out https://github.com/mattanger/ckb-next/

rainerborene, 2017-04-03:
ops. here's the right link:
[K70 Lux RGB ABNT 2, Cherry MX Red - CH-9101010-BR](http://www.kabum.com.br/produto/82067/teclado-corsair-gaming-mecanico-k70-lux-rgb-abnt-2-cherry-mx-red-ch-9101010-br/?tag=k70%20abnt%202)

rainerborene, 2017-04-08:
@tatokis do you know what should be done to make it work?

tatokis, 2017-04-08:
I've already implemented most of it, can't figure something out though.
What does the key next to the right shift do?

rainerborene, 2017-04-08:
@tatokis Here's the keyboard layout.
[image]

tatokis, 2017-04-16:
Perhaps I wasn't clear enough.
I was asking which key it maps to, when set to a a UK or ISO European layout.

tatokis, 2017-04-25:
Can you please open an issue in ckb-next? I keep forgetting about this.
And no, I meant the keyboard layout dropdown under the settings tab.