ccMSC/ckb #592: GNU/Linux - Ubuntu Installation problem

Tas-sos, 2017-04-11:
I have the Corsair K70 RGB and i want to customize the colors.
I made it through the windows from the official software ( http://www.corsair.com/en-us/downloads ) 
First of all, is not supported to save on the keyboard memory beyond static ( combination ) colors?
I was trying to save for example a Rainbow effects, but it was not possible.
Eventually i managed to save a fixed combination of illuminated buttons.

After that (and because i am Linux user - **only** - ), i tried to install on **GNU/Linux Ubuntu _12.04_ LTS**, operating system from this unofficial repository.

### **_First of all, you deserve many thanks and thumbs up for this effort!!_**

but, i have this problem :

```
 pwd
/home/user/Downloads/ckb-master
```

```
./quickinstall 
Preparing build files...
Compiling binaries...
(This can take a while, please be patient)
cd src/ckb-daemon/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && make -f Makefile 
cd src/ckb/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb/ckb.pro -o Makefile ) && make -f Makefile 
cd src/ckb-ripple/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb-ripple/ckb-ripple.pro -o Makefile ) && make -f Makefile 
cd src/ckb-wave/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb-wave/ckb-wave.pro -o Makefile ) && make -f Makefile 
cd src/ckb-gradient/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb-gradient/ckb-gradient.pro -o Makefile ) && make -f Makefile 
cd src/ckb-pinwheel/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb-pinwheel/ckb-pinwheel.pro -o Makefile ) && make -f Makefile 
cd src/ckb-random/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb-random/ckb-random.pro -o Makefile ) && make -f Makefile 
cd src/ckb-rain/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/user/Downloads/ckb-master/src/ckb-rain/ckb-rain.pro -o Makefile ) && make -f Makefile 
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-wave'
gcc -c -m64 -pipe -std=c99 -O2 -fPIE  -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o main.o main.c
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-daemon'
make -f Makefile.Release
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-ripple'
gcc -c -m64 -pipe -std=c99 -O2 -fPIE  -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o main.o main.c
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-gradient'
gcc -c -m64 -pipe -std=c99 -O2 -fPIE  -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o main.o main.c
make[2]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-daemon'
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/device.o device.c
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-rain'
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-pinwheel'
gcc -c -m64 -pipe -std=c99 -O2 -fPIE  -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o main.o main.c
gcc -c -m64 -pipe -std=c99 -O2 -fPIE  -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o main.o main.c
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb'
make -f Makefile.Release
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/devnode.o devnode.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/input_linux.o input_linux.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/input_mac.o input_mac.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/input.o input.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/main.o main.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/notify.o notify.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/usb_linux.o usb_linux.c
make[1]: Entering directory `/home/user/Downloads/ckb-master/src/ckb-random'
gcc -c -m64 -pipe -std=c99 -O2 -fPIE  -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o main.o main.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/usb_mac.o usb_mac.c
make[2]: Entering directory `/home/user/Downloads/ckb-master/src/ckb'
/usr/lib/x86_64-linux-gnu/qt5/bin/uic mainwindow.ui -o ui_mainwindow.h
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/usb.o usb.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/firmware.o firmware.c
/usr/lib/x86_64-linux-gnu/qt5/bin/uic kbwidget.ui -o ui_kbwidget.h
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/profile.o profile.c
/usr/lib/x86_64-linux-gnu/qt5/bin/uic settingswidget.ui -o ui_settingswidget.h
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/extra_mac.o extra_mac.c
/usr/lib/x86_64-linux-gnu/qt5/bin/uic kblightwidget.ui -o ui_kblightwidget.h
/usr/lib/x86_64-linux-gnu/qt5/bin/uic kbanimwidget.ui -o ui_kbanimwidget.h
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/keymap.o keymap.c
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -O2 -fPIE -DCKB_VERSION_STR=\"`cat /home/user/Downloads/ckb-master/src/ckb-daemon/../../VERSION`\" -I/usr/share/qt5/mkspecs/linux-g++-64 -I. -o release/command.o command.c
/usr/lib/x86_64-linux-gnu/qt5/bin/uic animadddialog.ui -o ui_animadddialog.h
/usr/lib/x86_64-linux-gnu/qt5/bin/uic animsettingdialog.ui -o ui_animsettingdialog.h
[... 146 lines cut ...]
                          ^
make[2]: *** [release/main.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from mainwindow.cpp:4:0:
mainwindow.h:60:26: error: ‘Qt::ApplicationState’ has not been declared
     void stateChange(Qt::ApplicationState state);
                          ^
mainwindow.cpp:276:34: error: variable or field ‘stateChange’ declared void
 void MainWindow::stateChange(Qt::ApplicationState state){
                                  ^
mainwindow.cpp:276:30: error: ‘ApplicationState’ is not a member of ‘Qt’
 void MainWindow::stateChange(Qt::ApplicationState state){
                              ^
make[2]: *** [release/mainwindow.o] Error 1
make[1]: *** [release] Error 2
make: *** [sub-src-ckb-make_first] Error 2

Press enter to exit.
```

Before run the "_quickinstall_" installation file, i run the following command:
`sudo apt-get install build-essential libudev-dev qt5-default zlib1g-dev libappindicator-dev`

So why are these problems?
I am disappointed that is not supported in Linux.  I did not expect that.

light2yellow, 2017-04-11:
`$ QT_SELECT=qt5 ./quickinstall`

ZirconiumX, 2017-04-11:
>First of all, is not supported to save on the keyboard memory beyond static ( combination ) colors?

The keyboards themselves don't support it. Colour effects are achieved by manually changing the colour of all the keys 60 times a second.

tatokis, 2017-04-11:
First of all, this project is abandoned, have a look at our fork. https://github.com/mattanger/ckb-next

Second, you are running Ubuntu 12.04, which never had Qt5, but is required for ckb.
If you added a PPA for it, then chances are the version you are using is either too old, or not properly installed.

I would suggest you upgrade to 16.04, which is currently the latest Ubuntu LTS.
12.04 is reaching End Of Life really soon (2017-04-28), so I do not think it is worth investing the time to make ckb run on it.

If you really do want to try and get it working, I suggest making sure you have a fully functional Qt5 installation, with all of the headers and the right paths being passed to the compiler.

If I am not mistaken, the oldest Qt5 version that ckb-next is known to work with, is 5.2.

But other than that, don't really expect support for a Ubuntu version that's about to be deprecated.

Tas-sos, 2017-04-11:
Thanks you all of them for so quickly response!

@light2yellow  Thanks you. If as you say set the version of Qt the process continues.

@ZirconiumX  , Thank you for your answer to that, it is important to me. You can be a little more detailed? I do not yet understand very well how it works and what functions it has this amazing keyboard. 

@tatokis Thanks, you too! Your answer is very important! I did not know that this repository - project is abandoned! They should inform the community so as not to confused the people.

light2yellow, 2017-04-11:
> They should inform the community so as not to confused the people.

README is meant to be read and tracked for changes. Look at it right now.

Tas-sos, 2017-04-11:
Oh! Yes you are right!  The mentions on top! Because of momentum behind did not notice!