ccMSC/ckb #603: Coirsair K95 Platinum no input when CKB is active

ShadowMemer, 2017-07-16:
When I first installed ckb I saw an firmware update notification so I clicked on update. 
But it was loading for a very long time so I just restarted my pc. (derp)
Then I looked between the other issues and saw that the update notifications were wrong. (Don't know if that has been fixed already)
When in Windows it says its up to date (according to CUE).
So when I now start ckb, my keyboard doesn't give any input. But when I close ckb and unplug and plug my keyboard back in it works again. My mouse (Corsair M65 Pro RBG) works fine when ckb is active. 
My keyboard works just fine in Windows. Also in Ubuntu but when ckb is not active.

Ubuntu version: 16.04  LTS
Keyboard: Corsair K95 Platinum
Mouse : Corsair M65 Pro RGB

ZirconiumX, 2017-11-25:
We're working on K95 Platinum support in [ckb-next](https://github.com/mattanger/ckb-next) - while not everything is perfect, you should have at least some basic support within the next few weeks.

rsparksm, 2017-11-25:
By basic support, do you mean functionality similar to Cue or just the ability to use the keyboard while ckb is running? Just picked up a k95 platinum and found the same issues as @ShadowMemer

ZirconiumX, 2017-11-25:
At the moment, "just the ability to use the keyboard while ckb is running". We're currently working on it.

If you want the technical reasons, the problem here is that ckb(-next) use USB URB Control packets, which the K95P just ignores, meaning CKB can't talk to the keyboard. Our plan is to use libusb for USB URB Interrupts, which *do* work.

mikeck1, 2018-06-23:
@ZirconiumX Still experiencing same issue 7 mo later. This issue is not resolved.

ZirconiumX, 2018-06-23:
This version of CKB does not support the K95P. CKB-Next does. Use that.