ccMSC/ckb #608: Causes lag online

CharlieD93, 2017-08-12:
Hi, I downloaded the driver and it made my mouse and keyboard work, but when I tested them on some games (League of Legends and Guild Wars 2) I found that it would make my ping vary between 30 up to 200, and would jump up in step of 10-15ms. I then uninstalled and the problem stopped.

joshliptzin, 2017-08-25:
I came here to mention this too. Here is a screenshot of pinging google.com with CKB open.

[image]

I had actually installed CKB a few weeks ago, hoping to use a corsair mouse that I liked better instead of my original mouse. I couldn't get it to work (forward/back buttons wouldn't work) and on top of that the mouse would randomly jump around the screen, so I went back to my old mouse. Forgot to uninstall CKB and only figured out it was the source of my frustrating ping spikes after several hours of debugging. I would not recommend anyone install this software at this time.

sunshine7913, 2017-09-23:
Yeah me too. Im having lags that I can not move my mouse properly. When I quit CKB, it works normally.