ccMSC/ckb #619: K65 RGB (NO devices connected) CKB est en version BETA 2.6

freasonlanders, 2017-12-15:
Bonjour je vient d'acheter le K65 RGB de chez corsair et le logiciel me marque NO DEVICES CONNECTED voila je ne sais pas quoi faire car les aides precedentes étaient de 2016 donc pas la meme version Merci de me répondre au plus vite si il y a encore des gens sur le forum

fleischie, 2017-12-15:
Bonjour, (excuser mon mouvaise français s'il vous plait).

Ce projet n'est plus maintenu. Essayez d'installer le projet [ckb-next](https://github.com/mattanger/ckb-next). Bonne chance! :smile:

freasonlanders, 2017-12-15:
he sa me donne c'est un fichier zip et sa me donne un dossier je fait comment ?

freasonlanders, 2017-12-15:
comment

freasonlanders, 2017-12-15:
comment     ?

freasonlanders, 2017-12-15:
commentt

freasonlanders, 2017-12-15:
ah sa me marque comment a lieu de coman

OvermindDL1, 2017-12-15:
This repository is inactive, you should proceed to the ckb-next repository that was linked in a prior post and acquire it instead as it is significantly more up to date.

And via Google Translate:

Ce référentiel est inactif, vous devez passer au référentiel ckb-next qui a été lié dans un post précédent et l'acquérir à la place car il est beaucoup plus à jour.

freasonlanders, 2017-12-16:
ah sa marche et comment on fait un macro de text ?

OvermindDL1, 2017-12-16:
As in this repository is old, go to https://github.com/mattanger/ckb-next instead.  :-)

And via Google Translate:

Comme ce référentiel est ancien, rendez-vous sur https://github.com/mattanger/ckb-next à la place. :-)