ccMSC/ckb #627: K70 layout problems on macos

matemaciek, 2018-04-03:
I have K70, ISO/European layout (small left shift, with `\|` next to it - let's call it the Problem Key). I'm using `ckb-next`.
- Choosing `English (ISO/European)` layout in ckb `Settings` results in **correct** layout in `Binding`, but the Problem Key works as `` `~``. Also `` `~`` key works as `§£`.
- Choosing `English (United States)` layout in ckb `Settings` results in **incorrect** layout in `Binding` (Long left shift, the Problem Key missing), the Problem Key works as `§£`, `` `~`` key works as it should.
So I 'm using `English (United States)` layout, but I guess it's not an expected behaviour.

ZirconiumX, 2018-04-03:
Can you try with [ckb-next](https://github.com/ckb-next/ckb-next) instead?
This branch says even in the readme that it's out of date.

matemaciek, 2018-04-03:
I'm actually using `ckb-next`, I'll search for similar issue there.

closed 2018-04-03 by matemaciek