ccMSC/ckb #633: Support for AIO's?

jbonnett92, 2018-10-08:
Hi,
Could you add support for AIO's please?

Thanks, Jamie

OvermindDL1, 2018-10-08:
What are AIO's?  Asynchronous I/O is the only definition of AIO that comes to mind for me?  o.O

Do you have a link to the corsair page for it?

OvermindDL1, 2018-10-08:
Also as an aside, this repository is dead, everything is on [ckb-next](https://github.com/ckb-next/ckb-next) now.

jbonnett92, 2018-10-08:
Ohh ok, it means All In One. More specifically the water coolers https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/c/Cor_Products_Cooling

OvermindDL1, 2018-10-08:
Ah those!  I would definitely say post that question on the `ckb-next` repo, that is where development has been going for years now and I bet they'd be up for it if you can help get them the information they need (USB packet capture and so forth, they can walk you through how).  :-)

jbonnett92, 2018-10-08:
Will do, I'm guessing I'd just need Wireshark?

OvermindDL1, 2018-10-08:
> Will do, I'm guessing I'd just need Wireshark?

That might work?  I use virtualbox with a windows guest to capture the windows packet information myself, but they would know for sure.  :-)

Their IRC channel is `#ckb-next` on freenode for note.