ccMSC/ckb #635: Suggestion: Allow macros for both keys being pressed and released

rabejens, 2018-12-14:
I am trying to configure some bindings on my Corsair Scimitar Pro for the music software Sunvox. There, you need to hold Shift while dragging to connect modules. It would be nice if I could, for example, bind `+lshift,+mouse1` to "Key 1 down" and `-mouse1,-lshift` to "Key 1 up".

If there is such a function, please tell me how to get there.

tdis8629, 2018-12-21:
Please use [ckb-next](https://github.com/ckb-next/ckb-next).

rabejens, 2018-12-21:
Sorry about confusing this, I "mis-clicked". I am actually using ckb-next. Does ckb-next have this feature or should I re-open this issue there?

tatokis, 2018-12-22:
It is not supported yet, however there is an issue open about it. You can subscribe to it.
https://github.com/ckb-next/ckb-next/issues/167