ccMSC/ckb #640: Mouse and keyboard do not work after installing

RBalconi, 2020-09-26:
Hello people,
I have the Ubuntu version 20.04 LTS and I am trying to install the ckb, however when I finish the installation none of my peripherals work. I have a K65 RGB keyboard and Saber RGB mouse. Even putting another generic keyboard / mouse does not work again, I had to format it a second time.
I am installing as requested in the Readme instructions:
- I ran the command `sudo apt-get install build-essential libudev-dev qt5-default zlib1g-dev libappindicator-dev` on the terminal.
- I downloaded the .zip and unzipped it.
- I ran `quickinstall` and gave` y (yes) `in all options

I also followed this video: [youtube video](https://www.youtube.com/watch?v=xfm2D15q4eE&ab_channel=laverdeparati)

Can someone help me?

Thank you

tatokis, 2020-09-26:
ckb is unmaintained. Install (ckb-next)[https://github.com/ckb-next/ckb-next/] from the PPA https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git

RBalconi, 2020-09-26:
> ckb is unmaintained. Install [ckb-next](https://github.com/ckb-next/ckb-next/) from the PPA https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git

Do I just execute these two commands in the terminal?
`sudo add-apt-repository ppa:tatokis/ckb-next-git`
`sudo apt-get update`

tatokis, 2020-09-26:
Yes, and then sudo apt install ckb-next.

RBalconi, 2020-09-26:
Just confirming:
`sudo add-apt-repository ppa:tatokis/ckb-next-git`
`sudo apt-get update`
`sudo apt install ckb-next`

In that order, right?

tatokis, 2020-09-26:
Yes

RBalconi, 2020-09-26:
> ckb is unmaintained. Install [ckb-next](https://github.com/ckb-next/ckb-next/) from the PPA https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git

Thank you very much, it worked. :)

closed 2020-09-26 by RBalconi