ccMSC/ckb #641: Corsair K95 key bindings

Daggerpov, 2020-10-16:
I'm aware that you have this listed as one of your "known issues" down below and that it says this: "Using the keyboard in BIOS mode prevents the media keys (including mute and volume wheel), as well as the K95's G-keys from working. This is a hardware limitation." but how exactly do I get around this limitation? I have a Corsair K95 (the one with 18 G-keys) and want to know how to bind them. For example, since I take French in school, I'd like to bind french accent keyboard shortcuts such as Alt-0224 like I would when using windows. Here's a snippet of what that looks like when I try doing so with your software:

[image]

Ravenslofty, 2020-10-16:
In BIOS mode the keyboard physically doesn't send this information to us.
To get it to work, switch out of BIOS mode to the 1ms mode.

But you should really be using ckb-next.

Daggerpov, 2020-10-16:
so how would I go about switching out of this mode?

Ravenslofty, 2020-10-16:
As far as I can tell from googling, there's a physical switch at the back
of the keyboard with options of BIOS, 1, 2 and 4. Just switch it to 1 and
it'll be fine.

Daggerpov, 2020-10-16:
The options are 8, 4, 2, 1 and it's already been set to 1.

Ravenslofty, 2020-10-16:
And your scroll lock isn't flashing? If it isn't, then you weren't in BIOS
mode to begin with.

Daggerpov, 2020-10-16:
So, there's still an issue I'm having whenever I try to set any key bindings involving the "Alt" character despite me not being in BIOS mode then.

Daggerpov, 2020-11-30:
The way I got around this, for anyone looking at this in the future is that I went into my system settings and turned on Compose Key in order to bind french accents as macro keys. 

Information about Compose Key: 

https://en.wikipedia.org/wiki/Compose_key#:~:text=A%20compose%20key%20(sometimes%20called,then%20n%20will%20insert%20%C3%B1.